weliem/blessed-android #192: With reduced number of UUIDs for scanning the sample app is no longer working.

crane-soft, 2023-11-08:
I successfully got the sample app running with my own Arduino based health temperature device using the HTS-SERVICE-UUID.
Then I tried to reduce peripheral scanning by deleting some UUIDs from the scan list in the startScann function:
```
    private void startScan() {
        handler.postDelayed(new Runnable() {
            @Override
            public void run() {
                central.scanForPeripheralsWithServices(new UUID[]{
                        HTS_SERVICE_UUID,   // scan only helth temperatur device
                        }
                );
            }
        },1000);
    }

```
But strangely the app no longer recognizes my temperature device. Whenever I delete just one UUID from the list, my device is no longer detected.

Attached my logcut:

```
2023-11-08 16:16:34.585  1483-1483  SDAgentPac...teReceiver system_server                        E  Not going to handle 'com.welie.blessedexample'!
2023-11-08 16:16:34.885  1483-1483  SDAgentPac...teReceiver system_server                        E  Not going to handle 'com.welie.blessedexample'!
2023-11-08 16:16:34.955  1483-1775  MARsDBManager           system_server                        E  insertPackage com.welie.blessedexample is already exist in DB!
2023-11-08 16:16:35.205  4026-4026  Launcher.Model          com.sec.android.app.launcher         E  onPackageChanged :com.welie.blessedexample
2023-11-08 16:16:35.795 11136-11136 ResourcesManager        com.welie.blessedexample             W  getTopLevelResources: /data/app/com.welie.blessedexample-1/base.apk / 1.0 running in com.welie.blessedexample rsrc of package null
2023-11-08 16:16:35.825 11136-11136 System                  com.welie.blessedexample             W  ClassLoader referenced unknown path: /data/app/com.welie.blessedexample-1/lib/arm
2023-11-08 16:16:35.955 11136-11136 ResourcesManager        com.welie.blessedexample             W  getTopLevelResources: /data/app/com.welie.blessedexample-1/base.apk / 1.0 running in com.welie.blessedexample rsrc of package null
2023-11-08 16:16:35.955 11136-11136 ResourcesManager        com.welie.blessedexample             W  getTopLevelResources: /data/app/com.welie.blessedexample-1/base.apk / 1.0 running in com.welie.blessedexample rsrc of package null
2023-11-08 16:16:35.955 11136-11136 AppCompatDelegate       com.welie.blessedexample             D  Checking for metadata for AppLocalesMetadataHolderService : Service not found
2023-11-08 16:16:35.955 11136-11136 ResourcesManager        com.welie.blessedexample             W  getTopLevelResources: /data/app/com.welie.blessedexample-1/base.apk / 1.0 running in com.welie.blessedexample rsrc of package null
2023-11-08 16:16:35.975 11136-11136 ResourcesManager        com.welie.blessedexample             W  getTopLevelResources: /data/app/com.welie.blessedexample-1/base.apk / 1.0 running in com.welie.blessedexample rsrc of package null
2023-11-08 16:16:35.985 11136-11136 art                     com.welie.blessedexample             W  Before Android 4.1, method android.graphics.PorterDuffColorFilter androidx.vectordrawable.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
2023-11-08 16:16:36.055 11136-11136 TextView                com.welie.blessedexample             D  setTypeface with style : 0
2023-11-08 16:16:36.055 11136-11136 TextView                com.welie.blessedexample             D  setTypeface with style : 0
2023-11-08 16:16:36.065 11136-11136 TextView                com.welie.blessedexample             D  setTypeface with style : 0
2023-11-08 16:16:36.125 11136-11136 SecWifiDisplayUtil      com.welie.blessedexample             D  Metadata value : none
2023-11-08 16:16:36.125 11136-11136 ViewRootImpl            com.welie.blessedexample             D  #1 mView = com.android.internal.policy.PhoneWindow$DecorView{71cfabb I.E...... R.....ID 0,0-0,0}
2023-11-08 16:16:36.135 11136-11183 OpenGLRenderer          com.welie.blessedexample             D  Use EGL_SWAP_BEHAVIOR_PRESERVED: true
2023-11-08 16:16:36.165 11136-11183 Adreno                  com.welie.blessedexample             I  QUALCOMM build                   : d842ad3, Ia10634f51b
                                                                                                    Build Date                       : 01/04/16
                                                                                                    OpenGL ES Shader Compiler Version: XE031.06.00.05
                                                                                                    Local Branch                     : mybranch17578993
                                                                                                    Remote Branch                    : quic/LA.BF.2.1.2_rb1.7
                                                                                                    Remote Branch                    : NONE
                                                                                                    Reconstruct Branch               : NOTHING
2023-11-08 16:16:36.165 11136-11183 libEGL                  com.welie.blessedexample             D  eglInitialize EGLDisplay = 0xb2c3f7c4
2023-11-08 16:16:36.165 11136-11183 OpenGLRenderer          com.welie.blessedexample             I  Initialized EGL, version 1.4
2023-11-08 16:16:36.195 11136-11136 ViewRootImpl            com.welie.blessedexample             D  MSG_RESIZED_REPORT: ci=Rect(0, 72 - 0, 0) vi=Rect(0, 72 - 0, 0) or=1
2023-11-08 16:16:36.315 11136-11136 Timeline                com.welie.blessedexample             I  Timeline: Activity_idle id: android.os.BinderProxy@c13f5f9 time:9289134
2023-11-08 16:16:36.645 11136-11197 Minikin                 com.welie.blessedexample             D  FontFamily bestFont == NULL, so return vacant FakedFont
2023-11-08 16:16:36.645 11136-11197 Minikin                 com.welie.blessedexample             D  FontFamily bestFont == NULL, so return vacant FakedFont
2023-11-08 16:16:36.645 11136-11197 Minikin                 com.welie.blessedexample             D  FontFamily bestFont == NULL, so return vacant FakedFont
2023-11-08 16:16:37.105 11136-11136 BluetoothAdapter        com.welie.blessedexample             D  STATE_ON
2023-11-08 16:16:37.105 11136-11136 BluetoothAdapter        com.welie.blessedexample             D  STATE_ON
2023-11-08 16:16:37.105 11136-11136 BluetoothLeScanner      com.welie.blessedexample             D  Start Scan
2023-11-08 16:16:37.105 11136-11136 BluetoothAdapter        com.welie.blessedexample             D  STATE_ON
2023-11-08 16:16:37.105 11136-11136 BluetoothAdapter        com.welie.blessedexample             D  STATE_ON
2023-11-08 16:16:37.105 11136-11136 BluetoothAdapter        com.welie.blessedexample             D  STATE_ON
2023-11-08 16:16:37.105 11136-11136 BluetoothAdapter        com.welie.blessedexample             D  STATE_ON
2023-11-08 16:16:37.155 11136-11147 BluetoothLeScanner      com.welie.blessedexample             D  onClientRegistered() - status=0 clientIf=10
2023-11-08 16:16:37.185 23106-23134 BtGatt.GattService      com.android.bluetooth                E  [GSIM LOG]: gsimLogHandler: com.welie.blessedexample, msg: MESSAGE_START_SCAN
2023-11-08 16:16:37.185 11136-11136 BluetoothCentralManager com.welie.blessedexample             I  scan started
2023-11-08 16:16:50.735 11136-11142 art                     com.welie.blessedexample             I  Debugger is no longer active

```

crane-soft, 2023-11-09:
I've made some more tests. When using the name of my device, its works:
`central.scanForPeripheralsWithNames(new String[]{"HELTH-Temper-Device"});`

weliem, 2023-11-12:
I just tried to reproduce your issue, but it seems to work fine on my phone. I used a thermometer I have lying around....

The most likely explanation of your issue is that you are not including the HTS service UUID in the advertisement. You can verify that with apps like NRF Connect.

crane-soft, 2023-11-13:
You are right, I can no longer reproduce the problem. I have made many changes to the sample app and it works perfectly.
I also tested your original sample app again and no problem either.
I have no idea what went wrong there.
Anyway you can close this issue.

closed 2023-11-14 by weliem