weliem/blessed-android #193: Drop in Sensor Data Frequency When Multiple Sensors Connected

michaellinder1995, 2024-01-07:
Hello,
I am encountering a problem with sensor data transmission frequency when multiple sensors are connected to an Android device (Xiaomi Note 12 Pro, android 13). The first sensor consistently transmits data at around 60Hz. However, when a second sensor is connected, the frequency of the second sensor drops to approximately 42-43Hz. This issue persists even when third or fourth sensors are added; their frequencies stay at around 42-43Hz, while the first sensor maintains its 60Hz frequency.  We have no problems on IOS devices. 

**Expected Behavior:**
Each sensor should consistently transmit data at its original frequency, around 60Hz, regardless of the number of connected sensors.

Actual Behavior:
With one sensor connected (first sensor): ~60Hz
With two or more sensors connected: First sensor remains at ~60Hz, while additional sensors drop to ~42-43Hz

I appreciate any insights, suggestions, or guidance you can provide. Thank you in advance for your help

michaellinder1995, 2024-01-08:
Update:
We tried two older Xiaomi phones (Xiamio Mi 9 Android 14, Xiaomi Redmi 11s Android 12) which worked fine. So its no issue of the lib i guess. Still would be nice to know whats the issue? Any specific hardware?

weliem, 2024-01-08:
Actual performance is influenced by a number of things, but none of them is this library. Things like the connection interval and BLE chip capabilities will influence the performance in your situation. For example, a BLE 5 chip may support a 2M Phy and hence be faster than a BLE 4 chip that only does 1M.

You could do a readPhy() to see what kind of connection you are on. Or look at the Connection Interval using the callback onConnectionStateUpdate.

Lastly, every chip has a limit on the maximum number of connections. Typically this is 5. But maybe the connection is already 'saturated' earlier because you are on 60 Hz per device....

closed 2024-01-09 by michaellinder1995