weliem/blessed-android #195: Best Practice for handling temporary disconnections?

jones139, 2024-04-12:
This is probably not an issue, more of a question, sorry!

I wondered what the best practice is for handling the situation where a peripheral goes out of range for a while, then comes back into range?   I do not seem to be getting an automatic reconnection even when I make sure the peripheral is awake - it is showing as not being connected to the phone.   Doing a complete re-start on the phone side restores the connection, but I would rather have the phone start trying to reconnect whenever the peripheral disconnects.  
I thought that was what the 'autoconnect' method was supposed to do, but it does not seem to work (Android 13).
Is there a best practice approach to doing this?
Thanks!
Graham.

weliem, 2024-04-25:
Issuing an 'autoconnect' should be sufficient. If that isn't working for you then please share relevant code and a log.

jones139, 2024-04-25:
Thanks for confirming.   It seems to be working ok now that I re-do autoconnect in the onDisconnectedPeripheral callback (my code is [here](https://github.com/OpenSeizureDetector/Android_Pebble_SD/blob/V4.2.x/app/src/main/java/uk/org/openseizuredetector/SdDataSourceBLE2.java#L200)).    I have an issue with the peripheral going to sleep if it is disconnected for too long, but I'll fix that in the watch firmware - thanks!

closed 2024-04-25 by jones139