weliem/blessed-android #196: No prompt for ACCESS_COARSE_LOCATION & ACCESS_FINE_LOCATION permissions

MarkusMiller, 2024-04-18:
Hello,

last year, I wrote a data-gathering app that takes images at fixed time intervals and gathers additional data from all sorts of APIs and sensors, such as GPS location. 

This year, I tried to add a wind sensor that is connected via Bluetooth and I finally managed to write a sandbox app, that successfully transferred data between the Bluetooth anemometer and the app using blessed-android.

I then tried to merge the old app with the new Bluetooth functionality but as it turns out, the GPS sensor doesn't work anymore due to missing permission. The Android system is not showing any prompt to grant `ACCESS_COARSE_LOCATION` and `ACCESS_FINE_LOCATION` and I spent a happy week watching sanity check after sanity check fail. Yesterday and today I started all over integrating part after part and as soon as I added 

`implementation("com.github.weliem:blessed-kotlin:3.0.6")`

to the gradle script, the permission prompt for GPS location stopped appearing when launching the app. Removing the line restored the behaviour. 

Unfortunately, I didn't find any documentation regarding the usage of the library along with GPS sensors. Is there any information on how to get both things working side by side, or is it maybe a bug?

Kind regards and thank you

jones139, 2024-04-18:
I had a lot of trouble with something similar a couple of weeks ago.   My issue was changes in the Bluetooth permissions between android 12 and android 13 (or maybe it was 11 and 12).   If your list of permissions us incorrect, it does not show the dialog to request permissions,  and fails quietly (I think this is an android bug but don't know where to report it).
Could including the library change your target or minimum sdk and therefore expose this issue for you?

weliem, 2024-04-18:
In later versions of Android, the location permissions are not needed anymore as they were replaced by dedicated BLE permissions. So if you need it for GPS functionality, you'll have to ask for them separately.

MarkusMiller, 2024-04-18:
Hello, thank you for getting back. 

@jones139 That were the permissions I actually tried to get (plus a couple of others) and the permissions for GPS were not granted. I just did a sanity check only granting the permissions you specified (line 102) and Android asked for permissions for all devices nearby but ACCESS_COARSE_LOCATION and ACCESS_FINE_LOCATION weren't granted.

@weliem I did a sanity check, only asking for `ACCESS_COARSE_LOCATION` and `ACCESS_FINE_LOCATION`, which worked perfectly fine, but as soon as I added the `implementation`-call above, Android stopped showing the grant permission prompt. With added GPS functions, those functions just stopped working as soon as I added the library, due to lacking permissions :-/

Any resource I could find on that topic told me to ask for COARSE/FINE permission but what am I supposed to do if I can't grant the permissions anymore, as soon as I add the implementation line?

MarkusMiller, 2024-04-18:
I'm not entirely sure, yet, but I think I might have found the issue:

In https://github.com/weliem/blessed-kotlin/blob/main/blessed/src/main/AndroidManifest.xml 

```
<uses-permission
        android:name="android.permission.ACCESS_FINE_LOCATION"
        android:maxSdkVersion="30" />
<uses-permission
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        android:maxSdkVersion="30" />
```

declare the permissions in question up to SDK version 30. I'm using version 33 as that's the version of my phone and as I just learned from a side note, Android merges manifests. So, by declaring the permissions in my manifest, the "missing" android:maxSdkVersion most likely got merged into my app's manifest once I added the implementation line. 

So, I added 

```
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"
        tools:node="remove"
        tools:selector="com.welie.blessed"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"
        tools:node="remove"
        tools:selector="com.welie.blessed"/>
```

into my manifest in the hopes that this would not merge the library's manifest into my app and as I just tested now, I'm getting asked by the system to grant location permissions and devices nearby permissions.

Not sure yet, if it actually works, but let's see...

weliem, 2024-04-21:
Yes, I think that should work!

MarkusMiller, 2024-04-21:
It's working, yes. Confirmed. I think this ticket can be closed now and the workaround for newer API versions could be included in the documentation.

closed 2024-11-17 by weliem