weliem/blessed-android #198: How to connect App to multiple ESP32

ashu2121, 2024-07-22:
I have done the testing successfully with one BLE device.

Please describe the steps to connect the library to multiple ESP32 devices at the same time
also i read that an Android phone can connect 3-9 ESP32 devices is it possible with your library too?
Do I need to have a separate BLE GATE declaration to connect the device?

Help will be highly appreciated.

weliem, 2024-07-22:
there is nothing special about connecting to multiple peripherals. Just scan and connect to peripherals you want to connect to....

ashu2121, 2024-07-23:
i checked the documentation of this library thoroughly and did not find anything about connecting multiple peripherals at the same time.
I m little confused as i m going over other tutorial, says i need multiple BLE gate object here is code preview 
```
central.autoConnectPeripheral(central.getPeripheral(MACADRESS, peripheralCallback);

 peripheralCallback = new BluetoothPeripheralCallback() {  ... }
```

do i need 
`BluetoothPeripheral` object for each peripherals to read and write `Characteristic`?
`BluetoothPeripheralCallback`  object for each peripherals? 

or is there anything else do i need to do in order to connect with multiple peripherals

it would be great a sample code snipet you  provide just  steps like i posted.

closed 2024-11-17 by weliem