weliem/blessed-android #200: Wait with responding?

satsen, 2024-10-24:
When running this as a peripheral (server), how can I send a response to a characteristic read later? After for instance user confirmation. The built-in android API uses BluetoothGattServer.sendResponse to send responses, but from what I can see this library needs a return value immediately.

weliem, 2024-10-25:
Why wait? This slows down communication speed.

satsen, 2024-10-25:
Hey @weliem, thank you for your reply. My use case is that when a characteristic is read, I want to first ask the user on the peripheral side whether he accepts that this characteristic will be read. So that if he denies it, another GattStatus is sent with no data but if he approves it it's a SUCCESS with data.

weliem, 2024-10-26:
Ok, a bit unusual use case I guess. The library doesn't support any delaying functionality. You can still block the thread until you got your user's approval.

satsen, 2024-10-29:
Ok, thank you. I am new to BLE so I'm not sure if it makes sense to do it this way but I think it's the easiest way.

satsen, 2024-10-30:
Seems like it's using the main thread so I can't block that. I guess I need to make a new thread for the BLE server then. Or maybe I should fork this and add support for deferred responses.

satsen, 2024-10-30:
Actually I started it in another thread but I guess android is calling blessed-android from the main (UI) thread because that's the thread I am running code in when logging from inside onCharacteristicRead, so I can't block it at all.

satsen, 2024-11-15:
In case anyone reads this in the future, there are timeouts that will affect you if you try to do this, so I guess it is better to use notifications or whatever to notify the central of the readiness. But a deferred response is still useful for some tasks that might need to be done, for example network requests on the peripheral end. Blocking the thread would freeze the app, so the changes I made are still useful. What I did was to introduce a ReadResponseInterface implemented by ReadResponse and DeferredReadResponse, and the onCharacteristicRead sets a function on the DeferredReadResponse to be executed when deferredReadResponse.respond(...) is called.

closed 2024-11-15 by satsen