weliem/blessed-android #201: PIN code is not asked for

satsen, 2024-11-05:
I'm running blessed-android as a peripheral, but other devices can connect to it freely without any confirmation even though they were not paired before. How can I disable this?

weliem, 2024-11-05:
You need to make one of the characteristics 'encrypted' for read and/or write

satsen, 2024-11-05:
@weliem Thank you for your reply. I did that, I changed `service.addCharacteristic(new BluetoothGattCharacteristic(uuid, PROPERTY_READ, PERMISSION_READ));` to `service.addCharacteristic(new BluetoothGattCharacteristic(uuid, PROPERTY_READ, PERMISSION_READ_ENCRYPTED));` but I could still connect without any confirmation. However it disconnected me when I tried to read the characteristic (blessed-bluez), REMOTE_USER_TERMINATED_CONNECTION.

closed 2024-11-10 by satsen